XarkLabs/PDQ_GFX_Libs #1: Why not to merge this library inside "Adafruit GFX"?

q2dg, 2015-05-27:
Thanks a lot

XarkLabs, 2015-05-27:
Hello,
This is a "re-mixed" replacement for Adafruit_GFX library designed for maximum speed on AVR MCUs (as used in Arduino etc.).  Adafruit is free to merge this code if they want, but I could see it being a support issue for them (and Adafruit_GFX supports other non-AVR MCUs, my focus was AVR  - after seeing Teensy 3.x get all the "speed-up" love).  The way this re-mixed library is designed (and one reason it is faster & smaller) required making fundamental changes, so would not be easy to "merge" these two programming "styles".
However, it was designed as "source compatible" so that if you are using Adafruit_GFX, with a few trivial changes to init, you can "drop-in" this PDQ_GFX library (and driver lib) for a nice speedup (on AVR with supported chipset).  I see it as an advantage to be able to choose between the two libraries, as Adafruit_GFX still has more drivers and more compatibility with other non-AVR parts.
That said, no doubt Adafruit could do a lot to improve the speed of their library with some techniques I use (they are free to "borrow").

Hope that helps answer your question.

q2dg, 2015-05-28:
Yes, definitely!!
Thanks a lot for your so well explained answer!
I close the issue, then

closed 2015-05-28 by q2dg

NicoHood, 2015-05-28:
Well they could add both versions. Doesnt c provide #ifdefs?
They really should have a look at your library, I aggree. If they ignore it, that is their problem.
q2dg you should probably open an issue at their github page, so they recognize it and maybe think about partly merging it, maybe as second option.

q2dg, 2015-05-28:
I've already done it, yeah!