XarkLabs/PDQ_GFX_Libs #3: Problems using this library with FastLED

thorstenweber83, 2016-04-24:
When trying to include FastLED in the same project i use PDQ_GFX.h and PDQ_ST7735.h in i get a lot of warnings and errors like this one:

```
In file included from /home/user/Arduino/libraries/PDQ_ST7735/PDQ_ST7735.h:62:0,
                 from /home/user/projects/arduino-project/arduino-project.ino:15:
/home/user/Arduino/libraries/PDQ_ST7735/PDQ_FastPin.h:268:52: error: previous definition of 'class FastPin<17u>'
 #define _DEFPIN_AVR(PIN, MASK, L) template<> class FastPin<PIN> : public _AVRPIN<PIN, MASK, _R(PORT ## L), _R(DDR ## L), _R(PIN ## L)> {};
```

I suspect the Problem lies in your Library already including FastPin.h, but im no expert in C/C++ matters. I could really use your help. Thank you :)

XarkLabs, 2016-04-24:
Hello,
Yes, no doubt there is a conflict when both libraries are included from the same source file.  I never considered a project using both FastLED and an LCD (D'oh).
One possible solution is to add a second ".ino" (or .cpp) file to your project and have one that use PDQ_GFX and another for FastLED (and don't "mix" them).
Another idea if you know a bit about programming  is to enclose all the PDQ_GFX stuff in a "namespace PDQ { ... PDQ includes here...  }" (since it is almost all in a header, mostly you mostly just need to include the headers inside the namespace and add a "using namespace PDQ" above your code).
Finally a simple, but brute force way to fix it is to "search and replace" for FastPin in the PDQ_GFX files and replace it with (e.g.) PDQFastPin so it won't conflict (I should probably do something like this for a future version of the library).
Hopefully one of these will work for you (let me know). :-)

freezedi, 2016-05-13:
i found for PDQ_ST7735  , swap() function is missed. I think its mean swapValue()

freezedi, 2016-05-13:
Some China TFT 1.44 128x128 SPI V1.1 REDTAB or  SPI V2.1 BLACKTAB has deffect to incorrect Y axis.
Choose in PDQ_graphicstest:

PDQ_ST7735_config.h
 #define ST7735_CHIPSET		ST7735_INITR_144GREENTAB

In file PDQ_ST7735.h
line ~397 changes
			//spiWrite16(x0+2);			// XSTART
			//spiWrite16_preCmd(x1+2);		// XEND
			spiWrite16(x0);			// XSTART
			spiWrite16_preCmd(x1);		// XEND

line ~413 changes
			//spiWrite16(y0+3);			// YSTART
			//spiWrite16_preCmd(y1+3);	 	// YEND
			spiWrite16(y0+32);			// YSTART
			spiWrite16_preCmd(y1+32);	 	// YEND
thanks for fast libs. very cool. I need to quality animate:) GL!!!

freezedi, 2016-05-14:
Hello. 
Whether the library will be adapted under esp8266 ? The interesting speed allowing to do full animation.

XarkLabs, 2016-05-16:
Freedi, thanks for the "Y axis fix".  I will add this as an option next chance I get.  This may explain a few other reports I got with ST7735 having issues.

As far as ESP-8266, that is an interesting idea, but this PDQ_GFX library is mostly focused on low-level AVR issues.  ESP-8266 can probably run original Adafruit version reasonable fast (I would think).

closed 2016-05-16 by XarkLabs