XarkLabs/PDQ_GFX_Libs #4: Does not work well with sd library

damarbo33, 2016-06-22:
Is there any chance to get sd and display working togheter at the same time?

damarbo33, 2016-06-23:
Just a clue! 

if I do tft.begin() and then SD.begin(SD_CS), I can't write nothing in the screen. 
But if I modify the method writeCommand in PDQ_ST7735.h to make initially 

**_SPI.beginTransaction(mySPISettings);_**

and int the end

**_SPI.endTransaction();_**

it starts screen working. The problem is SD still no working. 

Hope it helps

P.D: I am using standard arduino SD lib

XarkLabs, 2016-06-23:
Hello,
Thanks.  This is indeed useful and reinforces my suspicion that the Arduino SPI transaction changes "broke" sharing the SPI port.  I plan on investigating this issue this weekend (and fixing a few other small issues).

Andrewsolo, 2017-01-05:
Hello, I really want to use your excellent library, but can not adjust work with the SDcard. 
For example of issue you can try to change Adafruit`s "spitftbitmap" sketch (Adafruit_ILI9340\examples\spitftbitmap) for working with your lib. 
Adafruit`s lib working fine, but your not. 
I tried different libs for SD Card (Native "SD" lib and "SdFat")
Please help to fix this issue.