XarkLabs/PDQ_GFX_Libs #5: Doesn't work in multiple file projects

Pin666, 2016-07-26:
I wanted to replace my adafruit libraries in my current project, which consists of a .ino file and two .h and one cpp file. I replaced the includes of all adafruit libs with the PDQ_GFX libs but then I get compile errors. Btw. my project runs perfect with the adafruit lib but just to slow. Can't imagine what I'm doing wrong. Could you please be so kind and point me in the right direction? 

Thanks al ot!! 

Compiling 'VFDController' for 'Arduino/Genuino Uno'
Screen.h:In file included from
Screen.cpp:from
PDQ_ILI9340.h:In static member function 'static void PDQ_ILI9340::begin()
PDQ_ILI9340.h:627:2: error: 'SPI' was not declared in this scope
:SPI.begin()
PDQ_ILI9340.h:629:18: error: 'SPI_MODE0' was not declared in this scope
:SPI.setDataMode(SPI_MODE0)
PDQ_ILI9340.h:630:22: error: 'SPI_CLOCK_DIV2' was not declared in this scope
:SPI.setClockDivider(SPI_CLOCK_DIV2); \\ 8 MHz (full! speed!) [1 byte every 18 cycles]
Error compiling project sources

damarbo33, 2016-07-26:
Try to add 

`#include <SPI.h>`

or 

```
#if ARDUINO >= 100
 #include "Arduino.h"
 #include "SoftwareSerial.h"
#else
 #include "WProgram.h"
 #include "NewSoftSerial.h"
#endif
```

Pin666, 2016-07-26:
Thanks for your help.
I already tried that. No matter where (in what file) I include SPI.h. It doesn't work!
Here a small "project" to replicate my issue.

Thanks!

**File "PDQ_test.ino":**
```
#include <SPI.h>                                      
#include <PDQ_GFX.h>				// PDQ: Core graphics library
#include "PDQ_ILI9340_config.h"			// PDQ: ILI9340 pins and other setup for this sketch
#include <PDQ_ILI9340.h>			// PDQ: Hardware-specific driver library

#include "Screen.h"
Screen screen;

void setup()
{
  screen.begin();
}

void loop(void)
{
}
```
**File "PDQ_ILI9340_config.h":**
```
#define	ILI9340_CS_PIN		10
#define	ILI9340_DC_PIN		9
#define	ILI9340_RST_PIN		8
#define	ILI9340_SAVE_SPCR         0
```
**File "Screen.h":**
```
#include <PDQ_GFX.h>   
#include "PDQ_ILI9340_config.h"
#include <PDQ_ILI9340.h>

#include <Fonts/FreeSerif12pt7b.h>
#include <Fonts/FreeSans12pt7b.h>

PDQ_ILI9340 tft;

class Screen
{
  public: 
         void begin();
};
```

**File "Screen.cpp":**
```
#include "Screen.h"

void Screen::begin(){
  tft.begin();
}
```

Yona-Appletree, 2016-11-02:
The bottom line with this project is that it can only be used from a single cpp file in your project. Since Arduino uses the #include lines in the .ino file to setup library includes, you pretty much have to put the `#include` lines for the library there, and _nowhere else_.

Your issue is that you're trying to include the graphics library from Screen.h and then referencing that from both Screen.cpp and PDQ_test.ino. You need to move the includes to the ino, scrap Screen.cpp and put it's definitions in the ino file.

This really is a very major limitation of the library, one I've had some trouble with in writing libraries as well, and one I'd really like to see a good solution to.

XarkLabs, 2017-03-04:
It is possible to use multiple files, but all the PDQ_GFX code can only be used in one .cpp file (so I suggest putting all drawing functions in a single file and call those functions from the other files).  Including it in multiple files in the same "sketch" was not a design goal.

closed 2017-03-04 by XarkLabs