XarkLabs/PDQ_GFX_Libs #7: Compiler error.

gioreva, 2016-10-10:
Compiling:
\Arduino\libraries\PDQ_ST7735\examples\PDQ_graphicstest\PDQ_graphicstest.ino
I receive this eror:

Arduino:1.7.10 (Windows 7), Scheda:"Arduino Uno"

In file included from PDQ_graphicstest.ino:25:0:

D:\Documents\Arduino\libraries\PDQ_ST7735/PDQ_ST7735.h: In static member function 'static void PDQ_ST7735::drawLine(int, int, int, int, uint16_t)':

D:\Documents\Arduino\libraries\PDQ_ST7735/PDQ_ST7735.h:854:14: error: 'swap' was not declared in this scope

   swap(x0, y0);

              ^

D:\Documents\Arduino\libraries\PDQ_ST7735/PDQ_ST7735.h:860:14: error: 'swap' was not declared in this scope

   swap(x0, x1);

              ^

Errore durante la compilazione

  Questo report potrebbe essere più ricco
di informazioni con
  "Mostra un output dettagliato durante la compilazione"
  abilitato in "File > Impostazioni"

Yona-Appletree, 2016-11-02:
I submitted a pull request to fix this, or you can use my branch: https://github.com/Yona-Appletree/PDQ_GFX_Libs

XarkLabs, 2017-03-04:
The fix for this should be merged now.  Thanks,

closed 2017-03-04 by XarkLabs