XarkLabs/PDQ_GFX_Libs #11: ST7781 examples refiert to ST7735 and Uno/mega support

DeMysteriis, 2017-03-22:
Hi,

Is possible to use your libs with Arduino Uno / Mega? I can't get to use them properly.
Examples files refuse to compile because their refert to ST7735 that i didn't have installed, and when i modify the sketch by replacing ST7735 with ST7781, it's end to a white screen. 
I've see and tried the 2 modes in ST7781_Config, also the commented line about use of original adaGfx lib but i don't have any adafruit driver for st7781, just the mcu friends and the readme only report about ADAGFX installation. 
So All of my attempts, results to all kinds of compil error or white screen
Is there anything that i missed ?
Greetings!

samuelbles07, 2017-10-09:
I this to work too. Did you already did it?

XarkLabs, 2017-10-18:
Sorry, just needed a search and replace in the example source.  graphictest should compile now for ST7781.

closed 2017-10-18 by XarkLabs