XarkLabs/PDQ_GFX_Libs #19: How to use the library

ajax5692, 2019-05-12:
Hi,
I am new to Arduino and have a 1.8inch TFT LCD. I want to use the PDQ library, but I'm not able to install it somehow. Can someone please help me out? Mine has an LCD driver ST7735.

mrt-prodz, 2019-05-12:
It depends on which IDE you are using, if you are using the Arduino IDE as far as I can remember you should have an Arduino folder in your Documents (or home directory on Linux) and you should simply copy the folders PDQ_GFX and PDQ_ST7735 in there.

You should also copy PDQ_ST7735_config.h in your main project folder and edit it accordingly if necessary.

After that you can include the library and configuration file in your .ino project file :

```
#include <PDQ_GFX.h>
#include "PDQ_ST7735_config.h"
#include <PDQ_ST7735.h>
```

Try to compile and run the example provided for the ST7735 first (in examples folder PDQ_graphicstest), once it's working you know you have set the environment properly to run or make any other project using this library.

ajax5692, 2019-05-13:
Thanks for the response!

I have copied the folders PDQ_GFX and PDQ_ST7735 in the specified destination (snapshot included).
[image]

After that when I'm trying to compile the example provided for the ST7735, (in examples folder PDQ_graphicstest), it's showing something like this:
[image]

ajax5692, 2019-05-13:
Thanks for the response. It worked!

I'm trying to increase the frame rate for an image (black and white bars) in a 1.8inch LCD TFT, with the PDQ library, but with the present code, it does not seem to make a difference. Any help with that?

`#include <PDQ_GFX.h>
#include "PDQ_ST7735_config.h"
#include <PDQ_ST7735.h>
#include <SPI.h>
#include <TFT.h>            // Arduino TFT library

#define cs   10
#define dc   9
#define rst  8

TFT screen = TFT(cs, dc, rst);

int count=0;

void setup() {
  // initialize the screen
  screen.begin();

  // make the background black
  screen.background(0,0,0);

  // set the stroke color to white
  screen.stroke(255,255,255);

  // set the fill color to white
  screen.fill(255,255,255);

  #define BACKCOLOR 0xFFFF // White
  #define BLACK 0x0000 // Black

}

void loop() {
  if (count<1) {
  // draw a rectangle in the center of screen
  screen.fillRect(0, 50, 160, 25, BACKCOLOR);
  screen.fillRect(0, 0, 160, 25, BACKCOLOR);
  screen.fillRect(0, 100, 160, 25, BACKCOLOR);
  delay(2000);
  screen.fillRect(0, 50, 160, 25, BLACK);
  screen.fillRect(0, 0, 160, 25, BLACK);
  screen.fillRect(0, 100, 160, 25, BLACK);
  delay(1000);
  count=count+1;
  }
}`

The image that I need is this:
[image]

ajax5692, 2019-05-15:
Thanks for the response. It works really well now!

closed 2019-05-15 by ajax5692

XarkLabs, 2023-06-08:
> Hello!
>
> Thanks for sharing info how to get going with the PDQ_ST7735... I have tried the above on Arduino IDE 1.8.13 and it does not compile...
>
> I am on AT Tiny3216 and TFT SPI display with the ST7735s driver ...
>
> 1st I am missing the #include "PDQ_ST7735_config.h" For example, where can this file be found?
> 2nd if this uncommented, then error :
>
> #if !defined(ST7735_CHIPSET) || !defined(ST7735_CS_PIN) || !defined(ST7735_DC_PIN)
> #error Oops!  You need to #include "PDQ_ST7735_config.h" (modified with your pin configuration and options) from your sketch before #include "PDQ_ST7735.h".
> #endif
> Thanks in advance for any help!
>
> Best.

Hello,

It looks like all the examples were not tested for all devices (sorry).

However, the missing file mostly just defines the pins you are using to control the display.

So it looks like this file: https://github.com/XarkLabs/PDQ_GFX_Libs/blob/master/PDQ_ST7735/examples/PDQ_GFX_Button_test/PDQ_ILI9341_config.h

Can be replaced with: https://github.com/XarkLabs/PDQ_GFX_Libs/blob/master/PDQ_ST7735/examples/PDQ_graphicstest/PDQ_ST7735_config.h

(And check that the pins inside match your project).

Take it easy,
Xark