XarkLabs/PDQ_GFX_Libs #23: LCD White screen

Gabe-Brown, 2020-06-05:
I can't get this library to work with my ili9341 tft display. My screen works fine when using the code and circuit from here: https://educ8s.tv/arduino-2-8-ili9341-tutorial/ but the screen stays white when using the example code from this library. All pins are addressed the same and I have even tried code that just changes the screen colour with the two different libraries but it is the same. I am very new to Arduino so I might be doing something stupid but I think the problem might be a lack of logic level converter.
Any help would be appreciated.

XarkLabs, 2020-06-07:
Hello, it is difficult to know what your problem might be remotely (and many things might cause "white screen").  That module looks pretty standard.  Offhand I would suspect maybe not resetting the display properly, but that is just a guess (check that RST reset pin is correct in .h file with example).
No logic level conversion would be a problem (mostly for the display - it can harm it), but I don't understand why it would run with the other library.  Seems likely there is some other pin layout difference?

goa, 2021-02-02:
In my case, moving the RST pin to the Mega's RESET pin did the trick :)
Thanks for this wonderful piece of work @XarkLabs 🙏🙏🙏!

XarkLabs, 2021-02-02:
That is excellent! :slightly_smiling_face: I'm very glad it is working for you.  I appreciate the update, too.

closed 2021-02-02 by XarkLabs